Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #4: OSError: [Errno 121] Remote I/O error

lagarib, 2020-10-25:
Receiving this error after running `sudo python test.py Servo`

`Traceback (most recent call last):
  File "test.py", line 47, in <module>
    servo=Servo()
  File "/home/pi/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/Servo.py", line 6, in __init__
    self.pwm = Adafruit_PCA9685.PCA9685()   
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 322, in write_byte_data
    self._device.write(data)
OSError: [Errno 121] Remote I/O error`

ZhiJGuo, 2020-10-28:
@lagarib Please check if the connection board at the GPIO is installed correctly

simonsax, 2020-11-16:
I got the same error when I run the script whithout any batteries installed

ZhiJGuo, 2020-11-17:
@simonsax You have to give the robot battery, otherwise the chip on it will not work

closed 2020-11-23 by ZhiJGuo