Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #6: test.py RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

RussellTaylor83, 2020-12-26:
Hi all, 

got a kit for Christmas, running on a Pi4. All built and ready to test, and the following happens.

`pi@raspberrypi:~/git/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server $ sudo python test.py Led
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    led=Led()
  File "/home/pi/git/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/Led.py", line 22, in __init__
    self.strip.begin()
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault`

I will try to figure out how to fix it (looks like the rpi_ws281x library doesn't like Pi4s), but thought this should go here for others in the same position.

Thanks

Russ

RussellTaylor83, 2020-12-26:
Seems the issue has been raised at the library, nothing yet

https://github.com/jgarff/rpi_ws281x/issues/431

RussellTaylor83, 2020-12-26:
I'm writing a fix for this, I'll cover it here:

https://github.com/rpi-ws281x/rpi-ws281x-python/issues/56

DenzelChen, 2020-12-28:
Hi Russ,
Thank you for your feedback and solution.
It is very helpful for us and our users.
Have a nice day.
Sincerely,
Freenove

closed 2020-12-28 by DenzelChen