Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #8: Has no attributes

NotITNerd, 2021-01-02:
When I run Servo.py an error message pops up: "Adafruit_PCA9685" has no attributes "PCA9685". I tried everything I found nothing works.
[image]

ZhiJGuo, 2021-01-04:
@NotITNerd 
[image]
sudo pip3 install adafruit-pca9685

ZhiJGuo, 2021-01-04:
@NotITNerd  
Please check if your default Python is python3
[image]

ZhiJGuo, 2021-01-04:
@NotITNerd 
Please check if your default Python is python3
[image]
The module can be installed through "sudo pip3 install adafruit-pca9685"

NotITNerd, 2021-01-04:
It is python3 and adafruit-pca9685 is already installed.

DenzelChen, 2021-01-11:
Did you turn on the TWO switches? Could you please send me some pictures of the issue to support@freenove.com?

NotITNerd, 2021-01-20:
When I click the LOAD button with the power on it just shuts off. So I cannot have both buttons on with the power on. I think maybe that the PCB was soldered incorrect.

NotITNerd, 2021-01-20:
I will charge the battery overnight maybe they are just low.

closed 2021-01-27 by EthPan