Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #17: Power Adapter

ChR-iSz, 2022-04-10:
How i can use for tests, calibration, ect. a power Adapter instead accus ?

Which power adapter i need, and how i must connect.

The accus are very fast empty an i will use the dog for tests with power cable...

Any idea ?

DenzelChen, 2022-04-11:
You can use 12V，≥10A power adapter to try.

ChR-iSz, 2022-04-11:
Thanks. 

Really 12 Volt ? Booth accus are in sum 7,4Volt. Has the board a input tolerance from rb. +5 Volt ? 🤷‍♂️

DenzelChen, 2022-04-11:
Yes, the board can accept 12V.

ChR-iSz, 2022-04-11:
Hi Denzel,

Thats great news. Thanks.

Btw: The robot is amazing. Freenove has make a very good job here. Amazing documentation, very good quality and nice code snipped here on github. Top 👍

Then i try now connect a 12v power adapter on the batterie Pins and hope it will work. 

Thanks for your support, and have a nice day.

closed 2022-04-11 by ChR-iSz