Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #18: Can't get server to autostart on boot

Regenhardt, 2022-04-17:
Hi,

not sure why, hope you can help me. I followed the tutorial.pdf, but it won't automatically start.  
Manually calling `sudo python main.py -tn` starts the server and works as expected.

After following the tutorial exactly and it didn't work, I manually executed the start.sh and an error was displayed that it couldn't connect to display, which makes sense since I used ssh. So I added -tn to the call in the start.sh and now it works if I call it manually.

Since the point.txt isn't copied to ~ when rebooting, I guess start.sh isn't even being called, but I don't know why. 

Here is a screenshot of the actual file contents and `ls` output to show the executable status:

[image]

DiscipulusNakamoto, 2022-04-21:
4 days ago and zero help. my kit is missing hardware. all signs point to return to amazon asap and look elsewhere. thanks for posting this.

DenzelChen, 2022-04-22:
> 
Please refer to this video:
https://youtu.be/EWo_V3cXfNw?list=PLOkhax8xuWu3ldxhU4kDk0GDPlg9XWmKU&t=43

DenzelChen, 2022-04-22:
> 4 days ago and zero help. my kit is missing hardware. all signs point to return to amazon asap and look elsewhere. thanks for posting this.

You can also tell me the missing parts and your address. We can send you replacements.

Freenove Support Team
support@freenove.com

Regenhardt, 2022-04-23:
Oh so the server only autostarts when I login to the desktop? Can't I autostart it whenever I start the raspberry pi?

lacuna2, 2022-04-24:
so the instructions aren't clear.
Where you see ".../pi/..." you should replace with your username.
For me it was ".../Peter/...".

DenzelChen, 2022-04-24:
> Oh so the server only autostarts when I login to the desktop? Can't I autostart it whenever I start the raspberry pi?
It can autostart when I start the Raspberry Pi. Please check the video again.

lacuna2, 2022-04-24:
no it starts logged in or not

DenzelChen, 2022-04-24:
> no it starts logged in or not

It can start without login.

Regenhardt, 2022-04-27:
> so the instructions aren't clear. Where you see ".../pi/..." you should replace with your username. For me it was ".../Peter/...".

No worries, my user is indeed called "pi" (as you can see in the screenshot).

> no it starts logged in or not

When I follow the video exactly, the server only starts when I login to the desktop.   
Starting the raspberry pi does not start the server (legs aren't activated).  
Logging in using ssh also doesn't start the server.  
As soon as I login to the desktop via RDP/VNC the server starts and the legs are activated, just like in the video you linked earlier.

> It can start without login.

How? Do I need to add anything that isn't in the video you linked earlier?

DenzelChen, 2022-04-27:
Add -tn after sudo python main.py to try.

Add the following contents to “start.sh” file.
#!/bin/sh
cd "/home/pi/Freenove_Big_Hexapod_Robot_Kit_for_Raspberry_Pi/Code/Server"
pwd
sleep 10
sudo cp point.txt /home/pi
sudo python main.py **-tn**
Press Ctrl + O and then press Enter to save it. Press Ctrl+X to exit.

Regenhardt, 2022-04-27:
As mentioned in my original post, and visible in the screenshot aswell, I did just that - add -tn to the call so it doesn't open the window. Unfortunately nothing happens until I login via RDP/VNC.

To further test this I even tried adding `sudo python test.py buzzer` to the start.sh just before starting the main.py to have an additional cue, but that one didn't happen either. So I figured it must be the file being called at all, but since I usually don't use Linux I'm a bit lost here.

DenzelChen, 2022-04-27:
Can you use the commands to start server without window successfully?

1. Use cd command to enter directory where main.py is located:
_cd ~/Freenove_Big_Hexapod_Robot_Kit_for_Raspberry_Pi/Code/Server_
2. Run main.py:
_sudo python main.py -t -n_
or Run main,py with following command:
_sudo python main.py -tn_

Regenhardt, 2022-04-27:
Yes, this works perfectly fine. 

I can also login using ssh and then call `sudo ./start.sh` and the server starts as expected.

DenzelChen, 2022-04-28:
Sorry I don't know the reason. It works here.
I am using Raspberry Pi 4B and following OS:
https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2022-04-07/2022-04-04-raspios-bullseye-armhf-full.img.xz

DiscipulusNakamoto, 2022-10-11:
Thank you.Sent using the mobile mail appOn 4/21/22 at 8:59 PM, DenzelChen wrote:
                
            
            
                
                    From: "DenzelChen" ***@***.***>Date: April 21, 2022To: "Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi" ***@***.***>Cc: "Comment" ***@***.***>,"DiscipulusNakamoto" ***@***.***>Subject: Re: [Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi] Can't get server to autostart on boot (Issue #18)
                
                

Please refer to this video:
https://youtu.be/EWo_V3cXfNw?list=PLOkhax8xuWu3ldxhU4kDk0GDPlg9XWmKU&t=43