Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #21: Voltage calculations mismatched causing server failure

ShaneMazur, 2023-02-09:
Curious why these calculations don't match?

1. [Calculation in test module](https://github.com/Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/blob/a7fc66f9557b524a9717bcfedc9b041ce91e51ae/Code/Server/test.py#L78) `Power=adc.readAdc(0)/255.0*5.0*3`
2. [Calculation in ADC module](https://github.com/Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/blob/a7fc66f9557b524a9717bcfedc9b041ce91e51ae/Code/Server/ADS7830.py#L22) `battery_voltage=data[4]/255.0*5.0*2`

Im not an electrical engineer so I'm not sure they are even supposed to match but it was causing me to run into the message
`The batteries power are too low. Please recharge the batteries or replace batteries.` and then my server shutdown. Went out and bought new batteries - same thing.

Shaynee-MO, 2023-02-10:
[Additional supplement Tutorial.pdf](https://github.com/Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/files/10703421/Additional.supplement.Tutorial.pdf)
It is related to the board version. Please refer to the attachment to operate.

ShaneMazur, 2023-02-10:
Thanks @Shaynee-MO. Im on v1.6 so I'll proceed with the change. Just wondering, why for v2+ does it use 2 different calculations?

closed 2023-02-10 by ShaneMazur