Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #23: Can't get LED module to work

matsbror, 2024-02-20:
I have just assembled the robot dog. So far most things work as expected. However, not the LED module.
I have doubts when it comes to the wiring. The tutorial shows that the cable should be installed like this:

[image]

However, with the way that the connectors are mounted (see image below), the cable must be installed with the wires crossed.

[image]

So referring to the first image, the blue and yellow cables are reversed. 

Can this be rectified in software?

Shaynee-MO, 2024-02-20:
Hi!

If the cable is reversed, please correct the cable by exchanging the first and third cables on one side and try again.
[image]
[image]

Here is a video for your reference.
https://www.youtube.com/shorts/MJEGnXrQUAw

matsbror, 2024-02-20:
Thanks! I have the right cable order, but the way that the connector is mounted on the boards, they become reversed when connected. So I guess I need to make the cable reversed then.

matsbror, 2024-02-20:
I reversed the cable to be crossed and can confirm that the LED module now works. One of the connectors is probably mounted wrongly.

closed 2024-02-20 by matsbror