Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #25: The Windows client fails with File can't be opened for reading! in function 'cv::face::FaceRecognizer::read'

marcportabellaclotet-mt, 2024-06-04:
Running the Windows client fails

When running Freenove_Robot_Dog_Kit_for_Raspberry_Pi]/Application/windows/window.exe
(which I guess it should be named Client.exe),
fails with:

```
Traceback (most recent call last):
  File "Main.py", line 1088, in <module>
  File "Main.py", line 30, in __init__
  File "Client.py", line 16, in __init__
  File "Face.py", line 9, in __init__
cv2.error: OpenCV(4.5.5) D:\a\opencv-python\opencv-python\opencv_contrib\modules\face\src\facerec.cpp:61: error: (-2:Unspecified error) File can't be opened for reading! in function 'cv::face::FaceRecognizer::read'

[26052] Failed to execute script 'Main' due to unhandled exception!
```

Note: I don't have this folder `D:\a\opencv-python`. I guess this path is hard-coded in the executable.

Shaynee-MO, 2024-06-05:
Hi! 

Many thanks for contacting us.

The application is now named window.exe.

Please reinstall the resources and try again.
https://github.com/Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/archive/master.zip

If the issue remains, please follow the tutorial to install Python and related libraries, and run the client with commands to see if it works.

marcportabellaclotet-mt, 2024-06-06:
it works, thanks.