Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi #28: Failed to run the command "sudo python Servo.py"

SijieDingEsna, 2024-08-29:
When I run the command Servo.py. I get the following error:
Now servos will rotate to 90°.
If they have already been at 90°, nothing will be observed.
Please keep the program running when installing the servos.
After that, you can press ctrl-C to end the program.
Traceback (most recent call last):
  File "/home/pi/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/Servo.py", line 28, in <module>
    S=Servo()
      ^^^^^^^
  File "/home/pi/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/Servo.py", line 8, in __init__
    self.pwm = PCA9685(address=0x40, debug=True)   
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 32, in __init__
    self.write(self.__MODE1, 0x00)
  File "/home/pi/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 36, in write
    self.bus.write_byte_data(self.address, reg, value)
OSError: [Errno 5] Input/output error

I use the command "sudo i2cdetect -y 1" to get connected device information. It shows there is no valid device.
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 

I am sure I already enabled i2c and reboot system multiple times.
If you need more information, please let me know.

SijieDingEsna, 2024-09-01:
Please see the attached pic of my work. And I already put the batteries into the charger for over 2 nights. 
[image]
I don't connect any servos on the board. And I think it will not effect the result of i2cdetect. BTW I use wifi to connect raspberry pi with my laptop.

Shaynee-MO, 2024-09-02:
Please remove the Raspberry Pi and the mpu6050 chip from the robot board.

Connect the MPU6050 chip directly to the raspberry pi and run i2cdetect -y 1 to see if its i2c address can be detected.
[image]
[image]
[image]

SijieDingEsna, 2024-09-03:
The previous pictures make me confused. The first pic shows 
vcc -> right pin 2nd row
gnd -> right pin 3td row
scl -> left pint 2nd row
sda -> left pin 3td row

The second pic shows
vcc -> right pin 1st row
gnd -> right pin 2nd row
scl -> left pin 2nd row
sda -> left pin 1st row

Would you please tell me which connect method is right?

Shaynee-MO, 2024-09-04:
Sorry for the confusion it caused. It has to do with the angle of the shot.
[image]
[image]

SijieDingEsna, 2024-09-04:
I connect with mpu6050 as your guidance. But I still cannot get device by the command "i2cdetect -y 1" .
Please see my connection in the attached pic.
[image]

Shaynee-MO, 2024-09-05:
The issue may be on your Raspberry Pi.
Do you have other control boards such as Arduino and ESP32 to check?

SijieDingEsna, 2024-09-05:
I don't have other control board. In your mind which board is easier to do such test. I have poor knowledge about it. Thank you

Shaynee-MO, 2024-09-06:
You can buy an Arduino board.

BTW, you can connect the Raspberry Pi back to the robot board without the mpu6050 chip connected and run i2cdetect -y 1 again to see if it can detect the i2c addresses of other chips.